bwack/C64-Switchless-Multi-Kernal-27C256-adapter #2: Black screen at power on with C64C shortboard

kavanozz, 2021-01-28:
I built the shortboard for my C64C with EXROM enabled via TL866ii, programmed and installed 27C512 and configured at power on for my configuration (4th option). Kernal switching works by pressing Restore key for 2 seconds and Exrom reset works with 5 seconds pressing. But when I turn off the machine using power switch and turn on, I get black screen. If I press Restore key for 5 seconds the machine resets and boots properly and everything normal until I turn it off again. I repeated this many times and never could get the machine boot directly.

gh0stless, 2021-01-28:
I had problems with the shortboard in a C64-C. It didn't do a cold start. I changed lines 317, 334 and 343 the delay () value from 300 to 400, now it works.

kavanozz, 2021-02-04:
Thanks for the suggestion. I tried 400, then 600 but didn't fix my problem. I also tried with another Attiny85. Then something happened and my switcher stopped working completely. I will build another one and try again.

bwack, 2021-02-05:
I'd like to have a look at this problem soon. Is the cold start (quickly off then back on) causing the reset line generated by the 555 just too short ?
Suggestion, if you'd like to experiment:
in:
https://github.com/bwack/C64-Switchless-Multi-Kernal-27C256-adapter/blob/a58f48c7adf1041987e753e416d5cc985b262627/SKS64_SKETCH/SKS64_SKETCH.ino#L243-L257
Try changing next_state from STATE_IDLE to STATE_RESET. Then it should perform a reset on cold start.

1. Is your attiny85 from a trusted source? ebay/aliexpress bought IC's are almost all fakes.
2. Are your fuses set to: highbyte 0x62, lowbyte: 0x5f ?
3. are you using any cartridge device that also controls the reset line?

kavanozz, 2021-02-05:
It didn't matter whether a quick turn off/on or a long one. I will try code change once I build the new board.
1. All ICs from Mouser.com. I have 10 x Attiny85's. I have tried with 3 of them with the same results so far.
2. Yes
3. No, but I have C64 Lumafix installed. I didn't think to remove it when I was testing. I will try removing it with the new board before testing the code change.
BTW, I had tried a 27C256 with proper initialization (short-short led) and had the same results.

kavanozz, 2021-02-05:
I rebuilt a second board. Original code works with the same issue, but modified code doesn't work at all. Even reset with Restore key doesn't help. I can select the board/eprom type, or change the kernal bank. I can see it working via led responses but computer never boots. Also tested removing Lumafix which didn't make any difference.

I keep changing Attiny's and after a few swaps nothing works. I erase and reprogram and it sometimes helps. I used a total of 5 of them so far. Here is a question. Is it possible that if the board/eprom type is modified, the eprom selection resets to the first bank? I don't think this is the case because even after replacing Attiny and configuring it, I see it boots with 3rd or 4th kernal.

If I remove disable_reset control and reprogram using Arduino, is it Ok just to set the disable_reset bit using TL866ii, without rewriting flash and other portions?

bwack, 2021-02-06:
I'm sorry you have trouble. I don't have the time to look at it now, but I will be glad for anything you find.

1. The ATtiny's of yours are probably just fine. No need to swap them.
3. Lumafix works fine with the switcher.
I followed Sven's 250469 kernal/basic adapter, the nand-chip trick with having basic in bank0 and kernals in banks 1,2,... I haven't tested the C64C version.  Can you check Sven Petersen's trouble shoot here ? https://github.com/svenpetersen1965/C64-Keyboard-Controlled-Kernal-Switch/blob/master/Rev.%201/pdf/C64_Keyb_KSw_v1.pdf .. Pay close attention to contact continuity. Ensure that there is continuity from the socket to the motherboard, even better check from socket and over to the cpu to be sure there is no broken traces.

Regarding the question about eprom selection. I don't understand, sorry. Can you help me understand ? :) I have a suggestion though. You can remove the attiny and try fixing the address pins to say bank1. This should rule out the microcontroller. If the address lines are hardwired, and it still doesn't work then there must be something else wrong. 
[image]
At the ATtiny (I presume it is socketed) pulled out. Ground MCU_A14 and MCU_A15, and put 5V on MCU_A13. This should select bank1. (Ps. Selecting bank0 (basic) is "illegal" and is not supposed to work.)

Yes you can write the fuse bits without rewriting the code memory using the tl866ii

bwack, 2021-02-06:
Oh shoot, regarding the code changes. I see now that if the change is put in STATE_SETUP, then the reset will never be reached on a normal boot (not holding restore at boot). What i wanted to try was to reset the machine at power on. To do that you could try adding.

      pinMode(INTRST, OUTPUT);
      delay(300);
      pinMode(INTRST, INPUT);

at the beginning of loop()

StormTrooper, 2021-02-07:
I was having a different issue (also shortboard with 27C512) where the C64 didn't reset after selecting the ROM. 
You could select the ROM and then had to power off/on and then it booted correctly. 

I've now added the above code and its fixed it issue.  C64 resets correctly after selecting ROM

kavanozz, 2021-02-08:
I didn't have much time but worked on this a bit tonight. Here is my findings:
1. Adding the lines to the end of setup() also did not fix it for me (assuming I had to revert back the first change you suggested on "case STATE_SETUP").
2. I enable EXROM RESET functionality (0x62, 0x5F), I cannot do cold boot.
3. If I disable EXROM RESET functionality (0x62, 0xDF) with the original code, I cannot enter setup to select board/eprom type, **but it boots fine and I can change the kernal**. Since the default is longboard and 4 banks, it functions like that.

At the end, I changed the code to make shortboard/8 banks default and I can use it now.
  banks = EEPROM.read(BANKS_EEPROM_ADDRESS);
  if (banks==255) banks=8;
  shortboard = EEPROM.read(SHORTBOARD_EEPROM_ADDRESS);
  if(shortboard==255) shortboard=1;

I also have breadbin C64 with longboard 250407. I will build the SKS64 for it when I get a chance. Thanks for creating and supporting this device.

bwack, 2021-02-14:
Thanks ! That is an interesting find!
This suggest a couple of things I'd like to test. Especially 2. and 3. What is different with EXROM reset, and why does the machine need a reset from the attiny. Maybe the shortboards have really short poweron reset. Also I'd like to check why the attiny does not go into setup mode if EXROM reset is disabled (attiny reset pin enabled).

I will order parts and build the shortboard version and test it myself on my C64G. It has a 250469 board.
I wonder if I should make  a avrgcc project of the code and have a makefile produce all the variants (longboard,shortboard,romsize) automatically when executing make., but first I'd like to find the root cause.

kavanozz, 2021-02-15:
I built the longboard version and installed on my breadbin with 250407 rev A. Installed the original sketch code and used a 27C256 this time.  Disabled Attiny reset pin to enable EXROM. This one also doesn't go to setup mode, but default config is the right one. 
I removed Attiny, read it's config on TL866ii and confirmed the reset pin was disabled (0x62, 0x5F). I didn't have time to try another one, but I am now suspecting that these Attiny85's are peculiar, even though they come from a very trusted source. I will order different model (45 or 25 maybe) or from different source and try again when I have a chance.

I also installed RGB led with this one and it's beautiful. I decided to buy one for 64C. I found this item on Ebay. I think it should work.
https://www.ebay.com/itm/100pcs-2x5x7mm-4pin-Common-Cathode-Diffused-RGB-Tri-Color-Rectangular-LED-Diodes/121751750432

kavanozz, 2021-03-02:
Bwack, I finally got a bunch of Attiny85 and 45 from Jameco. But the behavior didn't change (even with suggested code changes from you and others).

One thing I need to correct is that, when I disable EXROM functionality, I can still enter setup mode. In my initial testing, my C64 machines were never booting when I entered setup mode. When I disabled EXROM function, C64 boots right away with whatever kernal config is present. That threw me off thinking it was not going into setup mode. But I realized, if I just kept holding Restore key, it was indeed going to setup mode. Now I can just use default code without modification. So part of my previous feedback was wrong. I updated that.

In summary, I cannot use the machines when I enable EXROM function by disabling Attiny reset pin.

bwack, 2021-05-02:
Hi Kavanozz. I have finally built one and installed it in my C64G with 250469 motherboard.
One thing I found is that I have same issue as you. I need a warm start after the cold start. I haven't tried disconnecting exrom yet.
Another thing I noticed was when I tried an EPROM (the ones with window) it was more stable than the EEPROM I have. Though problems with the EPROM too. I will hook up the scope and have a look. I'm suspecting a couple of things.
- the attiny doesn't boot fast enough compared to the longboard. Maybe this is a thing when you hardcoded the shortboard-7kernals mode and removing the call to eprom. Calling eprom probably takes some time..
- There is an and gate here.. It adds a propagation delay to the already delayed BASIC/EXROM signal from the PLA. Maybe in sum it is too much delay ?
I'll hook up my digital discovery logic analyzer and have a look at the timing soon.

Hans

bwack, 2021-05-03:
Good news.
While making a video I noticed that the LED was flashing to indicate EXROM_RESET mode.
Removing that delay at the beginning of the program helped the computer to warm start. I haven't digged any deeper, but flashing the led means changing A13..A15, and if the C64C has a shorter reset time than the C64 longboard, then it can be a problem :)
Here is the new firmware.
https://github.com/bwack/C64-Switchless-Multi-Kernal-27C256-adapter/releases/tag/fw1.3

kavanozz, 2021-05-26:
Hi bwack, thanks for the update and the fix. It's working for me without any issues now. Closing this issue. Great job!

closed 2021-05-26 by kavanozz