bwack/C64-Switchless-Multi-Kernal-27C256-adapter #3: Calrification on documentation (or bug?)

glbyers, 2021-03-19:
Hi,

Thanks for the project! I just wanterd to clarify something in the documentation regarding programming in HV programmer wrt reset disabled;

"Make sure that Fuse Low Byte, high byte are 0x62 and 0xDF if reset is enabled, or 0x62 and 0x5F if disabled."

I _think_ this statement is inverted (double negative). If I tick "RSTDISBL=0", I assume we're saying "enable reset", in which case high byte will be 0x5F. If we leave "RSTDISBL=0" unticked, this would be "disable reset". Is that right? Thanks!

glbyers, 2021-03-20:
I looked at the doco for the attiny and I guess this is why you made the point to say high byte should be 0x5f if disabled. I think the TL866-II interface is confusing. I'll close this one. Thanks!

closed 2021-03-20 by glbyers

bwack, 2021-03-21:
Thanks! You were right about why I only mention the hex values. The tools are many and it can get confusing very fast.