bwack/C64-Switchless-Multi-Kernal-27C256-adapter #4: No power led when Eprom set in 27C512 (shortboard)

lolof70, 2021-05-01:
Hello,

All is working but when using (short board version) with a 27C512 (windbond) in 27C512 mod, the power led is not on.
It just light on when switching kernel or in setting mode.

Using the same 27C512 but with setting as 27C256 (in setting mode), I just have 3 kernel but power led is working as it should...  What could I try to get the led power on on 512 mode ?

Thank you.

darylrichards, 2021-05-01:
No, the same pin for A15 on the ROM is used for the LED. It can't be both.

lolof70, 2021-05-01:
Thanks, I did not knew. Then, I will switch to 256 Mod.
Thanks a lot for this very nice switcher

lolof70, 2021-05-01:
After some try, I have massiv problems....
If using alone without cartridge it's switching well between the kernel.
But if it's in use with cartridge like EF3, KFF or with a sidekick64, this does not work as it should I think.
When killing the EF3 or KFF, it does not return to the basic, black screen with lot of corruption most of the times...
When using with Sidekick64, when returning in Menu I can see for a short time that before to go into the sidekick64 menu it calls a kernel that is not the active one then freeze, seems it always calls the last kernel from the eprom, in this case Evox....
I tried in 512 as 256 mode, when 256 with 512 eprom I doubled the content of the eprom...
I did the bin file in this order :
64c.251913-01 + Jiffy + Evox...  (33k) I doubled it to fill the 27C512. I then try with the setting for short board 256 eprom and also short board 512 eprom... Results is the same.

I did correctly filled the Eprom ?
What could I try to get the sks64 working nicely with those cartridges.

Thanks again for the help.

bwack, 2021-05-02:
Did you try to go to basic with the menu button on the KFF? I tested it with my long board version. If already in basic and switching kernals machine returned to basic. But with the KFF in menu mode, it crashes when kernal switching.. Then I can click menu and I get back to menu.

The sks64 is dead simple. To the C64 it issues a reset and provides ROM set to a bank. Thats it. How can I fix this I dont know.

Have you tried disconnecting exrom from the sks64 ?

lolof70, 2021-05-02:
Thanks for the reply.
With Exrom line connected, when going to Basic from KFF (F7) = crash.
Just tried with Exrom line disconnected, a quick test show that it solves the issues I have. 
With sidekick64 is starting (when calling its menu), it  does not show anymore another kernel than the one active from sks64.
Same with KFF, no crash when going to basic, switching kernel....

The fuse are set like this with my TL866 : Low Byte 0x62 and high Byte 0xDF. I just read that using Exrom, reset should be not active and for that High Byte should be 0X5F. 

Is it correct ? Can it be the mistake ?

bwack, 2021-05-02:
Hi. I'm working on this now and soldering up a board. It is not a mistake. In the user guide i say that for reset disable you need 0x5F. I think maybe the confusion that many has is that it says #DISBLRST=0.. Theres like three negations there so it looks confusing. A checkmark in minipro means that the bit is 0, and it is active zero. so to get DISBLRST feature you need a check mark there (0x5F). :)

lolof70, 2021-05-02:
Then I did this mistake... I have to set it 0X5F for Exrom....
Yes, mistake is from me, not from user guide... When I told "can it be the mistake ?", I meant with this "my mistake" to have set the fuse to 0x62...

I will try with 0x5F and will tell you if it's working fine.

Thanks

lolof70, 2021-05-02:
With Exrom connected and with the fuse set to 0x5F, it's working stable from my first test.
I have the same as you with my KFF. Changing kernel in KFF is working but I have to do a reset after. It's probably due to the way the KFF is working. I still have to experiment deeper. But this solved also issues I had with sidekick64...
Perfect ! :-)

darylrichards, 2021-05-02:
Just to ask - are you running the version with the EXROM init patch I submitted? Otherwise EXROM is held high instead of floating...

lolof70, 2021-05-02:
After some more test.... Not stable. It's almost the same. It works once than after power off on, almost same as before :-(
I am using the hex version 1.1 from 25 July.

bwack, 2021-05-02:
Thanks Daryl. I'll update the hex files on the releases file.

bwack, 2021-05-02:
New release.
Going to test this now.
https://github.com/bwack/C64-Switchless-Multi-Kernal-27C256-adapter/releases/tag/FW_1.1

lolof70, 2021-05-02:
Did some test with 1.2

With expander KFF active, I can go to basic then switch to jiffydos for exemple than the C64 freeze till I push the KFF bouton menu. From there I can quit and go to Jiffydos. From Sidekick64/EF3, impossible to quit to basic. It freezes the C64

Thought , I am not sure I am doing all right...

bwack, 2021-05-02:
I tried the new firmware myself today. I experience problems too with cold boot. I also noticed that it was almost impossible with a winbond EEPROM (reprogramable/without window), and it worked better with aother EPROM (an eprom that has window). Will have to crack out the logic analyzer soon :)

bwack, 2021-05-03:
Good news.
While making a video I noticed that the LED was flashing to indicate EXROM_RESET mode.
Removing that delay at the beginning of the program helped the computer to warm start. I haven't digged any deeper, but flashing the led means changing A13..A15, and if the C64C has a shorter reset time than the C64 longboard, then it can be a problem :)
Here is the new firmware.
https://github.com/bwack/C64-Switchless-Multi-Kernal-27C256-adapter/releases/tag/fw1.3

bwack, 2021-05-03:
While it got better, sometimes i need to powercycle the machine to get to the KFF menu. I don't know how to fix that.

lolof70, 2021-05-03:
Just tried new firmware on long and short motherboard.

I did quick tests with cartridges alone. No expander....

Short board :

-Without cartridges, it seems to work well, I just had one time a problem by switching kernel...

- KFF, that's better now, the C64 start and I can select kernels. Just the same as before, if I switch kernel, I have to go again in KFF menu and F8 to quit to basic with new kernel. Switching from Basic freeze the screen till I go again in KFF with menu button. With exos V3 Kernel, it does not work.... Normal kernel, jiffy ok.

- Sidekick64 start and go directly in Sidekick menu as it should. When quitting to basic after to have started a game, it freeze. I can go back in sidekick menu and work with the sidekick but I never can go back to basic. I have to power the c64 few seconds, start in sidekick menu and quit to basic (without starting anything before to quit). 

- EF3, That's the cartridge who is working the best. Nothing special...

All of those test were done quickly.... Maybe there will be other things...

On lang board, it' almost the same problems as on the short one.

Thank you for all the work on this. I can imagine this is a nightmare with the timing with all those expansions cartridges....