bwack/C64-Switchless-Multi-Kernal-27C256-adapter #5: Longboard 8 Kernal doesn't set A15 high after cold boot

Neukam, 2021-07-19:
I'm using the SKS in a 1541 with the A14 and A15 address lines tied to the drive's device ID inputs so that I can switch between JiffyDOS and stock as well as setting the device ID.  I'm using SKS64_SKETCH_V1_3_attiny85_RGB_LED.hex firmware set in Longboard 8 kernal mode.  If I power cycle the drive with kernal 4-7 selected, the SKS will not set A15 high at power on.  A13 and A14 will return to their previous state as they should, but A15 will always be low.

bwack, 2021-07-26:
How do you control the attiny ? The attiny relies on a momentary switch (the restore key) as input.

bwack, 2021-07-26:
I'm looking through the code for possible issues.. I just wonder if its got to do with the fact that the same led was originally used for blinking the red LED.

Neukam, 2021-07-26:
I installed a small tactile switch in the front of the case with hot glue and connected it to the restore input and ground.  It works great except that if A15 is high when I turn it off, it will always be low when I turn the power back on.  The same behavior can be seen on my C64 if I set it to 8 kernal mode, but I normally have that in 4 kernal mode so it isn't a problem for me.
[image]

bwack, 2021-07-26:
The mod looks perfect!
I'm finding some uninitialized variables that may cause this. I've seen this problem now and then without knowing why, then it would just seem to work.
I wonder, does the drive have a connection with the reset line of the c64 ? I wonder if I could make some crude communication over the reset line. When the SKS64 in the c64 holds the reset, the SKS in the 1541 could listen for a message on the reset line such that the C64 and 1541 has the same rom number.

Neukam, 2021-07-26:
The drive does not reset if I hit a reset button on a user port or cartridge port device.  It does reset when the C64 is power cycled and if I send @I in JiffyDOS.    I'm looking at the schematics now to see if it would be possible.  It would be an interesting project.  It might be useful to set the state of A13 to select JiffyDOS or stock depending on which rom is selected on the C64 with the SKS sketch programed to know which of the available ROMs need JiffyDOS.  I'm more interested in seeing the A15 issue fixed though as I almost always just use JiffyDOS.

Neukam, 2021-07-26:
On 326298 and 250407 boards the serial bus reset is tied to the internal reset circuit.  On 250425 and 250466 boards, the serial reset is the EXTRST signal buffered by Q3 and which is separated from the internal reset circuit by U8.

Neukam, 2021-07-26:
It's also important to note that the 326298 boards have a different reset circuit than all the later boards and can't just be pulled low without modifying the reset circuit first.

bwack, 2021-07-27:
I have the sks64 on a breadboard, and I see the same issue as you see. A15 is always low at power on in 8 bank longboard mode.

bwack, 2021-07-27:
I released a new firmware that fixes this bug. The A15 bit was not stored on eprom.
https://github.com/bwack/C64-Switchless-Multi-Kernal-27C256-adapter/releases/tag/fw1.4

Neukam, 2021-07-27:
Thanks!  I've tested the ATTINY45 and ATTINY85 RGB precompiled versions and they both work as expected.  If you or anyone else wants to put one in a 1541, I recommend reducing PRESSTIME to 8 and HOLDTIME to 18 to make changes and resets faster.

bwack, 2021-07-30:
I added serial data to the reset signal, and a listener on the same line :) WIll try this in a C64 and a 1541 drive soon:)
https://twitter.com/bwack77/status/1420889235939278848?s=20