bwack/C64-Switchless-Multi-Kernal-27C256-adapter #9: Unable to get 1.6 working

StormTrooper, 2022-02-05:
I've got a longboard 250425 and using a attiny85 with a single LED. 

Version 1.4 works fine, but I have now tried to upgrade to 1.6 and its programmed fine, but I cannot it working. 

On bootup I hold the restore in  and it does not go into setup mode. Doing the same on 1.4 works fine. 

Are there any hardware changes that need to be done between 1.4 and 1.6?

hansdals, 2022-02-08:
No.
You have: the longboard version, and have flashed the SKS64_SKETCH_fw_1.6_RED_attiny85.hex firmware ?
I haven't tested with a single led. Not sure why it should be a problem. When I have time I can have a look, unless someone else here beats me to it :)

bwack, 2022-02-08:
No.
You have: the longboard version of the sks, and have flashed the SKS64_SKETCH_fw_1.6_RED_attiny85.hex firmware ?
I haven't tested with a single led. Not sure why it should be a problem. When I have time I can have a look, unless someone else here beats me to it :)

StormTrooper, 2022-02-08:
Yes, I flashed SKS64_SKETCH_fw_1.6_RED_attiny85.hex and used on a longboard.  Haven't managed to do any other testing yet on a shortboard or with RGB led yet.

AtomicRPM, 2023-04-15:
Works for me.  Using a PIC12F629 though.

kevinherzig, 2024-03-10:
I know this was opened long ago, but an issue on newly built boards for me too.  Have to use 1.4 to get this to work.  Will take a look at the code.