bwack/C64-Switchless-Multi-Kernal-27C256-adapter #12: Problem on 250469 not all kernels are switched

danko79, 2022-10-01:
Hi, i have a C64c mainboard short 250469 and flashed SKS64_SKETCH_fw_1.6_RGB_attiny85.hex on attiny85-20pu with TL866II programmer and have pcb short rev 1.1 with flash W27c512-45z flashed the 7 kernals plus basic, BASIC_KERNAL_EXOS_SPEEDDOS_JIFFYDOS_JAFFYDOS_DOLPHIN_SPEEDDOS.bin the problem is the multi kernal adapter switched only the last 4 kernal the others banks not switched, i can't figure out where i am wrong should 7 kernal work?

bwack, 2022-10-02:
Here are the instructions from the User Guide about choosing long/short board and how many banks. Befor you power on the machine, you hold restore while you power up the machine. After 5 seconds of holding restore you should see a white flash. The speed of the flashes indicate configuration. see below.

```
Setup
After installing the firmware in the SKS64, and installing it in the computer you can enter the setup menu
by holding RESTORE while powering on the machine. Continue to hold RESTORE until the LED flashes white.
Tap RESTORE to rotate through all the four possible setups. Two flashes, short or long, can be
seen periodically. They indicate the selected setup where the first flash is the motherboard form factor, and
the second flash is related to the size of the EPROM used.
	LONG – SHORT	longboard, 4 banks
	LONG – LONG	longboard, 8 banks
	SHORT – SHORT	shortboard, 3 banks
	SHORT – LONG	shortboard, 7 banks.
	LOONG		1541 mode, 2 banks.
Hold RESTORE again for 5s to store setup in the ATtiny's EEPROM, and the main program starts.
```

danko79, 2022-10-02:
Yes works! Many Thanks.

closed 2022-10-02 by danko79